oresat/oresat-adcs-hardware #2: Delete 'Board Schematic PDFs' folder

NathanDus, 2018-06-15:
Hi Eric, I don't know how to cleanly remove things from Git. It doesn't like it when I try. Can you delete the folder 'Board Schematic PDFs'? Its not needed anymore. We have things in the 'Board Schematic PDFs and Images' folder.

heliochronix, 2018-06-15:
@NathanDus,
To remove something from Git, use `git rm`. It acts pretty much like regular `rm` but also stages the changes inside git. For example, in the oresat-acs-board directory you would do `git rm -r Board\ Schematic\ PDFs`. Hope that helps!

heliochronix, 2018-06-15:
Also, then you would `git commit` the changes. Note that empty directories are automatically removed from git repositories if there's nothing inside.

closed 2018-06-17 by E-Ruhl