oresat/oresat-adcs-hardware #4: OreSat Standard Card Mount

E-Ruhl, 2018-06-17:
This needs to be developed in parallel with the ACS driver board mounts so that the connections can easily be made between the boards and the card

Joedang, 2018-06-22:
I don't think the ACS driver board will have any effect on the card mounting per se. I still think it's a good idea to have a flexible connection between the  ACS driver boards and the card. That wouldn't mess with the way the card goes into the structure. At any rate, we should meet at some point to play with the CAD and figure out a good form factor for the Rev4 driver board.

E-Ruhl, 2019-06-29:
Currently looking into FFC connections. Potential problem with mechanically fastening them. Will talk with Patrick about potentially modifying cage to operate as a clamp.

andrewgreenberg, 2022-10-11:
This is fixed in ADCS >= v1.2 and upcoming RW rev 7.

closed 2022-10-11 by andrewgreenberg